mne-tools/mne-python #13121: Consider reverting lazy imports

cbrnr, 2025-02-20:
I would like to discuss reverting the changes that introduced the `lazy-loader` package. I have been unhappy with this change basically almost from the start (although I made a positive comment in the initial discussion, but this was before I fully understood how lazy loading was implemented), and I think it is not good for the project's health in the long run for the following reasons:

1. Lazy loading is essentially a sophisticated workaround rather than a supported Python feature. The relevant [PEP 690](https://peps.python.org/pep-0690/) was rejected, and the [related discussion](https://discuss.python.org/t/pep-690-lazy-imports-again/19661/26) suggests it's unlikely to be officially adopted in the future. As a result, the mechanism is not guaranteed to work in all cases and has already shown instances of breaking.
2. The package relies on `.pyi` files for runtime behavior, which contradicts their intended purpose. Official references, such as [PEP 561](https://peps.python.org/pep-0561/#definition-of-terms) and [PEP 484](https://peps.python.org/pep-0484/#stub-files), explicitly state that stub files are strictly meant for static type checking and not execution.
3. Lazy loading can be useful for CLI tools or packages with many submodules, where startup time is a major concern. However, I think the advantages for MNE-Python are questionable:
    - The import time wasn't particularly long to begin with.
    - While a 50% reduction in relative terms sounds significant, the absolute gain is just [175 milliseconds](https://github.com/mne-tools/mne-python/pull/11838#issue-1825317577), which is negligible.
    - We already used nested imports for optional dependencies, and nested imports can also be used to import packages on demand without the need for the non-standard `lazy-loader` mechanism.
    - Delaying imports until first use might be more disruptive than a slightly longer initial import time.
4. The `lazy-loader` package does not appear to be very actively maintained beyond some tooling updates, and there are several unresolved issues such as all packages being loaded at once ([issue #131](https://github.com/scientific-python/lazy-loader/issues/131)) and eager imports not working as expected ([issue #128](https://github.com/scientific-python/lazy-loader/issues/128).

Given these concerns, I strongly suggest that we at least reconsider our decision to adopt the `lazy-loader` package. Since this will be a rather large change involving many files and probably a lot of manual work, I'm happy to submit a PR if there is general agreement that this is the right direction. It is also not super urgent, but I think the sooner we address this, the easier it will be to revert the changes.

drammock, 2025-02-21:
This comment has the relevant backstory links that @cbrnr forgot to include:

https://github.com/mne-tools/mne-python/issues/12388#issuecomment-1910876635

> we started with https://github.com/mne-tools/mne-python/pull/11838 (but did it a dumb way, without .pyi files). This caused problems with IDE completion hints etc, as discussed in https://github.com/mne-tools/mne-python/pull/12059 (that's where all the opinions are aired). The IDE problem was fixed by https://github.com/mne-tools/mne-python/pull/12072.

cbrnr, 2025-02-21:
It's not the entire backstory, just some aspect. I think I summarized all important points in my initial post.

drammock, 2025-02-21:
> It's not the entire backstory, just some aspect.

Sure. I wasn't trying to imply that your description wasn't relevant. Just that there are some other things that are also relevant that you didn't include.

> I think I summarized all important points in my initial post.

I disagree.  Past discussions / decisions are relevant and important context. I'm surprised that you seem to be saying that you omitted them *intentionally*.

cbrnr, 2025-02-21:
Of course I'm not saying I omitted something intentionally, come on!

drammock, 2025-02-21:
> Of course I'm not saying I omitted something intentionally, come on!

I guess I misunderstood. Saying "I think I summarized all important points in my initial post" implied that you didn't think linking to past discussions was important, so you chose not to do it. Which, again, was surprising to me, especially since it's a frequent request when old decisions get revisited --- I assumed you'd have thought of doing so.

mscheltienne, 2025-02-23:
I agree with @cbrnr I don't see much value in `lazy-loading` and I'm +1 to revert it.

hoechenberger, 2025-02-23:
> 4\. The `lazy-loader` package does not appear to be very actively maintained beyond some tooling updates, and there are several unresolved issues such as all packages being loaded at once ([issue [...]

I find this concerning and I never liked the fact that we now critically depend upon a hack that didn't pass standardization (via a PEP) for IMHO very good reasons. If forced to vote, I'd vote against the current lazy loader implementation.

hoechenberger, 2025-02-23:
FWIW I just briefly checked some of the projects that are listed as having "endorsed" SPEC-0001: NumPy, SciPy, xarray.

In **none** of the above could I find traces of the `lazy-loader` dependency!? Either I'm missing something or the way the SPEC website presents itself is kind of misleading.

larsoner, 2025-02-24:
I still like the lazy loading stuff but can live with the majority decision here. I guess we'll see what problems/harder other decisions come out in #13124 -- one we've already hit that people should hopefully discuss is how to handle the `sklearn` dependency in `mne.decoding`: https://github.com/mne-tools/mne-python/pull/13124#issuecomment-2678596013

larsoner, 2025-02-24:
> In none of the above could I find traces of the lazy-loader dependency!? Either I'm missing something or the way the SPEC website presents itself is kind of misleading, suggesting adoption where there is none. Happy to be corrected, of course, but right now I'm kind of … confused.

FWIW just looking into one, SciPy uses a lazy loading approach even if it's not via `lazy_loader`, see https://github.com/scipy/scipy/pull/15230 / https://github.com/scipy/scipy/blob/b9b8b8171fd1453b42fc4492a279c71b54141c51/scipy/__init__.py#L129 . I think for DRY/community adoption it would be better to use `lazy_loader` but I haven't looked into why they use their own code. I'm assuming the same would probably hold for NumPy, xarray, etc.

cbrnr, 2025-02-24:
No objection against nesting the slow imports as we previously did (including `sklearn`).

larsoner, 2025-02-24:
sklearn isn't as simple, copying from https://github.com/mne-tools/mne-python/pull/13124#issuecomment-2678596013 so we can discuss in one place :

For `sklearn` we have a harder decision. Without lazy loading `mne.decoding` we basically have two choices:

1. Require people to import `mne.decoding` before accessing anything in that namespace. So you could no longer do:
   ```
   import mne
   mne.decidng.CSP(...)
   ```
   You would need to add a `import mne.decoding` instead (or `from mne.decoding import CSP` etc.). A lot of other packages are this way, such as `sklearn` (and SciPy was at some point, might still be).
2. Make `sklearn` a hard requirement of MNE.

I guess I'd vote for (1). It's a bit weird if we do it just for one submodule, but I guess it's justified because it's the only one that requires an optional dependency for its class hierarchy.

Beyond that I'm not sure how we'd get classes in `mne.decoding` to properly subclass `sklearn` classes (e.g., `TransformerMixin`, `MetaEstimatorMixin`, etc.). Before lazy loading we had our own duplicated, half-baked, partially incorrect variants of these subclasses, which we definitely should not go back to. It's a big maintenance burden and constantly breaking in addition to being a bad violation of DRY, not working with `sklearn` class validation tools, etc.

hoechenberger, 2025-02-24:
> Before lazy loading we had our own duplicated, half-baked, partially incorrect variants of these subclasses, which we definitely should not go back to.

+100

drammock, 2025-02-24:
Well, the ship may have already sailed, but I'd like to take the time to enumerate some arguments on *both* sides of this issue.

#### in favor of keeping lazy loader

1. **`import mne` is faster.** Even if it's less than 200 ms, abandoning it now will be a performance regression. In a single interactive session you might not notice, but over dozens of CI runs per day (many of which we pay for) those seconds will add up. MNE import time also impacts import time of all downstream projects that themselves import MNE.
2. **we get the benefit of nesting expensive dependency imports without actually having to nest them.**  Previously there were many PRs where a contributor failed to nest a SciPy import, which caused a test failure, which often they didn't know how to interpret (or even access) meaning maintainer time was spent explaining, coaching, and/or fixing the problem. With lazy_loader, there's no longer a need to next external dependencies, which aligns with contributors' default expectations.
3. **It's not causing us any problems.** MNE-Python isn't experiencing any of [the problems cited by @cbrnr](https://github.com/mne-tools/mne-python/issues/13121#issue-2866709198):
    - https://github.com/scientific-python/lazy-loader/issues/131 is an edge case that doesn't affect us: it behaves suboptimally when dependency packages aren't found (AKA, when your environment has problems, it doesn't work perfectly).
    - https://github.com/scientific-python/lazy-loader/issues/128 looks quite similar to something we hit, and fixed. It was a genuine problem with MNE that was revealed by lazy_loader adoption, not a lazy_loader bug.
    Since #12072, things have been working smoothly for us AFAICT.
4. **Less churn for contributors.** As mentioned in (2) above, our import preference / policy now aligns with what people want/expect to do anyway (put external imports at the top of the file). It's a simpler system than what we had before (only *some* external imports got nested), and for folks who have been contributing in the last few years, I worry that we'll look a bit scattered (and contributors will be confused) going from nested to non-nested and back to nested in the span of less than 2 years.

#### against keeping lazy loader
(I've kept these in the same order as @cbrnr's [original post](https://github.com/mne-tools/mne-python/issues/13121#issue-2866709198)

1. It's not a built-in feature of Python (and likely never will be).  
2. It uses `.pyi` files for purposes they weren't intended for.
3. The speedup doesn't matter. "Delaying imports until first use might be more disruptive than a slightly longer initial import time."
4. lazy_loader is not actively maintained.

#### response to arguments against keeping it

> It's not a built-in feature of Python.

I can't fathom why this should hold any weight. Lots of things will never make it into the standard library.

> It uses `.pyi` files for purposes they weren't intended for

This is the only argument I find somewhat convincing. It seems unlikely, but it is possible that Python will someday set stricter rules about `.pyi` files that would interfere with them being used in this way. More concerning to me, though, is what might happen when we decide to make MNE fully type-annotated, and decide that we want/need the `.pyi` files for, e.g., multiple dispatch definitions.

> The speedup doesn't matter

See point (1) in favor of keeping lazy_loader.

> lazy_loader is not actively maintained

Maybe. I'd say it's a bit too early to tell. In https://github.com/scientific-python/lazy-loader/issues/128 Stéfan said "ping me if you don't get it sorted out" and he was never pinged again, so I feel it's unfair to count that as evidence for lack of maintenance.  There's also a bit of a built-in backstop, in that `lazy_loader` is housed within https://github.com/scientific-python which has a lot of very competent people who could step in / take over if Stéfan ever did go missing.

cbrnr, 2025-02-25:
Just one quick response to the point on CI runs: if we switch from `pip` to `uv pip` (not even plain `uv`), we get the following install times for `mne[dev]` (with cached downloads of course):

- `pip install "mne[dev]"`: 22s
- `uv pip install "mne[dev]"`: 0.35s

So about 62x speedup, which I think makes arguing about 200ms longer import times a moot point.

hoechenberger, 2025-02-25:
Marginally (?) related but has anyone considered that the metrics we're using here are not really useful anyway? Just comparing `import mne` times is pointless; one needs to measure the cumulated import times until everything is loaded that I'll need to perform a certain action or analysis. Making everything import "lazily", i.e., deferring most imports to "later", and then just omitting them in the measurements is probably not a valid approach.

cbrnr, 2025-02-25:
> To me this is a non-starter. Interactive API exploration is something I use a lot with both MNE (where sometimes I forget the exact name of a func, so I mne.viz.<TAB> to find it), or just to explore what an unfamiliar package offers.

I don't use it at all, I guess we will not be able to come up with a statistic on how the majority of users require that feature or not. In addition, you can still interactively explore if you first `import mne.viz` (to stick with your example, it would be sufficient to just not export the largest/slowest modules to the `mne` namespace).

drammock, 2025-02-25:
> you can still interactively explore if you first `import mne.viz` 

which is *even slower* than importing everything eagerly in the first place.

> (to stick with your example, it would be sufficient to just not export the largest/slowest modules to the `mne` namespace).

which breaks interactive API exploration. How is that "sufficient" exactly?