mne-tools/mne-python #13133: getpwdcmd error in mne.bem.make_scalp_surfaces

darcywaller, 2025-02-27:
I am receiving an error when running the make_scalp_surfaces function in a Jupyter notebook. The following command
_mne.bem.make_scalp_surfaces(subject=subject,subjects_dir=subjects_dir,mri='T1.mgz')_

outputs
_1. Creating a dense scalp tessellation with mkheadsurf...
Running subprocess: mkheadsurf -subjid sub-104 -srcvol T1.mgz -thresh1 20 -thresh2 20
getpwdcmd: Command not found._

with the verbose error log displaying an error in run_subprocess that leads to
_CalledProcessError: Command '['mkheadsurf', '-subjid', 'sub-104', '-srcvol', 'T1.mgz', '-thresh1', '20', '-thresh2', '20']' returned non-zero exit status 1._.

Freesurfer is loaded with FREESURFER_HOME defined and subject and subjects_dir are certainly defined and do not generate errors when running mne.bem.make_flash_bem in the previous cell. 

I'm running MNE version 1.8.0 and freesurfer version 7.3.2. Could you please advise what might be the issue?

larsoner, 2025-02-27:
Can you successfully run `mkheadsurf --help` on your system?

darcywaller, 2025-02-27:
Yes! mkheadsurf --help is successfully returning the function documentation in the terminal and with ! in my jupyter notebook.

shresth-keshari, 2025-02-28:
@darcywaller You're right that FREESURFER_HOME is set correctly, especially since mne.bem.make_flash_bem is working. However, make_scalp_surfaces specifically relies on mkheadsurf, which isn't used in make_flash_bem.

Try running the command directly in a terminal:
`mkheadsurf -subjid sub-104 -srcvol T1.mgz -thresh1 20 -thresh2 20
`
If it works here but fails in Jupyter, the issue is with Jupyter’s environment.
If it fails here too, FreeSurfer itself may be misconfigured.

darcywaller, 2025-02-28:
Running that full mkheadsurf command in both the terminal and the jupyter notebook still produces
_getpwdcmd: Command not found_.

I'm assuming this is a freesurfer bug and can work with my local IT team or freesurfer support (it's a Linux module I cannot change the version of) to try to resolve.

darcywaller, 2025-03-25:
Thanks all - I addressed this issue by working with our Linux IT team to make sure the folder including getpwdcmd in the Freesurfer package is loaded with the Linux module, which wasn't happening.

closed 2025-03-25 by darcywaller