mne-tools/mne-python #13139: BUG: Parallel build JSON file issues

larsoner, 2025-03-03:
We get doc build errors like this fairly frequently nowadays:

https://app.circleci.com/pipelines/github/mne-tools/mne-python/26824/workflows/2f0db808-f2fd-4931-841c-ac7c7d98da4f/jobs/71898

<details>
<summary>Traceback</summary>

```
    Unexpected failing examples (1):
    
        ../examples/preprocessing/otp.py failed leaving traceback:
    
        Traceback (most recent call last):
          File "/home/circleci/project/examples/preprocessing/otp.py", line 81, in <module>
            bias = compute_bias(raw)
                   ^^^^^^^^^^^^^^^^^
...
          File "/home/circleci/project/mne/parallel.py", line 95, in parallel_func
            cache_dir = get_config("MNE_CACHE_DIR", None)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
          File "/home/circleci/project/mne/utils/config.py", line 309, in get_config
            config = _load_config(config_path)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^
          File "/home/circleci/project/mne/utils/config.py", line 234, in _load_config
            warn(msg)
          File "/home/circleci/project/mne/utils/_logging.py", line 416, in warn
            warnings.warn_explicit(
        RuntimeWarning: The MNE-Python config file (/home/circleci/.mne/mne-python.json) is not a valid JSON file and might be corrupted
```

</details>

This started happening once we enabled parallel building in docs. I suspect it could be because the JSON file is being written by one example as it's trying to be read by another (or maybe more likely, written by one Python thread while another tries to read it). There are a few potential solutions:

1. Ignore the corrupt-JSON warning in doc builds. A bit ugly and makes it so we'll miss if something more serious goes wrong with our config I/O. But I'll probably open a PR to do this for now just so that the build stops failing so often.
2. Stop building docs in parallel. Has the severe disadvantage of doubling our build time.
3. Use some file-lock mechanism to prevent reading while writing (or writing while writing).

It seems like (3) is probably the best solution. We could use `filelock` (it's used for example by `virtualenv` and maintained by `tox-dev` so should be well supported; license is public domain) but it would require adding another hard dependency. We could also use [msvcrt](https://docs.python.org/3/library/msvcrt.html#msvcrt.locking) and [fnctl](https://docs.python.org/3/library/fcntl.html#fcntl.flock) but we'll be reinventing the wheel. Another option would be to vendor some or all of their code.

drammock, 2025-03-03:
> We could use `filelock` [...] but it would require adding another hard dependency.

Only in the `doc` optional dependency list though right?

> We could also use [msvcrt](https://docs.python.org/3/library/msvcrt.html#msvcrt.locking) and [fnctl](https://docs.python.org/3/library/fcntl.html#fcntl.flock) but we'll be reinventing the wheel

[This SO answer](https://stackoverflow.com/a/46407326) is only ~50 LOC, so if that's representative of what it would take to reinvent the wheel, seems maybe worth it to facilitate sticking to the standard library.

larsoner, 2025-03-03:
> Only in the doc optional dependency list though right?

No we'd need to ideally use it any time we read from or write to the config JSON, and also the [forward dot product cache files](https://github.com/mne-tools/mne-python/blob/266ccc9540f159609976fc6e1b2259c93923b2e3/mne/forward/_lead_dots.py#L53).

> This SO answer ...

+1 for trying that. I can take a stab at it hopefully soon and we can see if it works okay in our `main` doc builds

bruAristimunha, 2025-04-09:
Hey @drammock and @larsoner,

We are facing the same problem with moabb all the time. 

I was testing some way around here: 
https://github.com/NeuroTechX/moabb/compare/develop...bruAristimunha:moabb:try-to-solve-parallel

But no success. Also, re-writing the get_config and set_config functions from mne is not a good idea. Any input is welcome!

larsoner, 2025-04-09:
Actually I thought of a hopefully fairly easy intermediate solution... make FileLock an *optional* rather than *required* dependency. Then we don't have to maintain our own code for file locking. Any time we `get_config` or `set_config` we lock the config file (if we can), as well as any time we try to write the forward dot products. Like a new `mne/utils/config.py` function like
```
@contextlib.contextmanager
def _open_lock(path, *args, **kwargs):
    filelock = _soft_import("filelock", raise_error=False)
    if filelock is None:
        ctx = contextlib.nullcontext()
    else:
        ctx = filelock.FileLock(...)
    with open(path, *args, **kwargs) as fid, ctx:
        yield fid
```
and in code we just use `with _open_lock(fname, "w"):` or similar.

And I suggest that someday if this becomes a more widespread problem (outside of doc builds and development!) then we can roll our own file locking solution based on the StackOverflow issue.

drammock, 2025-04-09:
nice idea!

bruAristimunha, 2025-05-05:
Hey @drammock and @larsoner,

Just to give you some feedback about this solution. It seems to work with internal tests and at moabb level, but in my case, the amount of time that you look at the file is critical, 5 seconds seems to be best.

https://github.com/NeuroTechX/moabb/pull/749/files#diff-df9901d2cd27f2a6b7a28fdea331ab6699aaab31500d30af57aadfc82b4c8454R11-R46

Do you want me to push this into the mne? I think it will potentially solve the issue.

larsoner, 2025-05-05:
> Just to give you some feedback about this solution. It seems to work with internal tests and at moabb level, but in my case, the amount of time that you look at the file is critical, 5 seconds seems to be best.

Not totally sure what you mean, ideally `.mne/mne-python.json` would be open for a tiny fraction of time (like in the tens of ms at most?) and not held open. Same with the other files like the forward cache file...

But in any case, yes, PR welcome for optional `filelock` use

bruAristimunha, 2025-05-06:
opened @larsoner and @drammock

PR #13241 (2025-05-21): FIX: Parallel build JSON file issues 

Fixes #13139.

#### What does this implement/fix?

Implementing a small semaphore to ensure we are always writing mne config json correctly even if we have parallel file access. Using filelock as optional dependency.

#### Additional information

This is affect moabb and braindecode a lot.
closed 2025-05-21 by larsoner