mne-tools/mne-python #13151: BCI2000 .dat file reader

Sam54000, 2025-03-10:
### Describe the new feature or enhancement

Hi, I have worked with a lot of [BCI2000](https://www.bci2000.org/mediawiki/index.php/Main_Page) output data and noticed that there is no existing reader in MNE (as far as I know). BCI2000 is a very versatile software for multimodal acquisition and the most cited BCI tool. Therefore, I would like to contribute to the codebase by adding a Raw class dedicated to BCI2000 data and its associated reader for output files in `.dat` format. I hope this is relevant, and I really look forward to contributing to the codebase!

### Describe your proposed implementation

Reproducing the same implementation philosophy of other format (such `RawEDF`, `RawEGI`) I would like to add a Raw class dedicated to BCI2000 data that would be called `RawBCI2k` and the associated reader `mne.io.read_raw_bci2k()`.
The BCI2000 community had already implemented a reader in python ([see here](https://www.bci2000.org/mediawiki/index.php/BCPy2000)) so I would like to take example of their reader and wrap it into a nice `mne.io.read_raw_bci2k()` function.
In order to avoid dependencies, I am going to write the class and the reader "from scratch" taking example from their reader and cite them.

### Describe possible alternatives

As an alternative, I would work with the BCI2000 team in order to make their reader a python package. We would put the BCI2000 reader in the mne requirement (it would be installed alongside the other packages when installing mne). Then we would call their reader module when calling `mne.io.read_raw_bci2k()`.

larsoner, 2025-03-11:
Sounds reasonable to me!

One hiccup could be licensing -- if you want to look at their code and understand what to do their code needs to be in a BSD-compatible license. Looking at https://github.com/neurotechcenter/BCpy2000 it looks like GPL, which is incompatible. You would need to ask for explicit permission to relicense their code.

If they don't grant it, you would have to come up with an implementation from scratch based on some file format specification doc or similar I think

Sam54000, 2025-03-13:
Sure, I am working with them to find the best solution.
Then I will implement the solution and submit the PR for review.

HansujaB, 2026-02-27:
@tsbinns Got it , thanks a lot !
Can you review my PR as well?