mne-tools/mne-python #13154: Trouble with measurement dates when anonymizing or writing epochs data

SophieHerbst, 2025-03-11:
### Description of the problem

Hello,

Not sure if both problems are related, but I am getting errors about measurement dates when trying to anonymize or write epochs data that was created with the mne-bids-pipeline in the last months. I would rather avoid having to run the pipeline again, as I just wanted to quickly take an example dataset, anonymize it, and use it for teaching.

**1) When trying to anonymize the data, I get the following error:**

```
epochs = epochs.anonymize()
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[59], [line 1](vscode-notebook-cell:?execution_count=59&line=1)
----> [1](vscode-notebook-cell:?execution_count=59&line=1) epochs = epochs.anonymize()

File <decorator-gen-26>:12, in anonymize(self, daysback, keep_his, verbose)

File ~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/_fiff/meas_info.py:795, in SetChannelsMixin.anonymize(self, daysback, keep_his, verbose)
    [793](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/_fiff/meas_info.py:793) info = self if isinstance(self, Info) else self.info
    [794](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/_fiff/meas_info.py:794) anonymize_info(info, daysback=daysback, keep_his=keep_his, verbose=verbose)
--> [795](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/_fiff/meas_info.py:795) self.set_meas_date(info["meas_date"])  # unify annot update
    [796](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/_fiff/meas_info.py:796) return self

File ~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/_fiff/meas_info.py:851, in SetChannelsMixin.set_meas_date(self, meas_date)
    [848](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/_fiff/meas_info.py:848)             value["machid"] = _tmp
    [850](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/_fiff/meas_info.py:850) if hasattr(self, "annotations"):
--> [851](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/_fiff/meas_info.py:851)     self.annotations._orig_time = meas_date
    [852](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/_fiff/meas_info.py:852) return self

AttributeError: 'NoneType' object has no attribute '_orig_time'
```

**2. When trying to save the data without anonymization, I get this one:**

```
epochs.save('example_data_epochs.fif', overwrite=True)
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[61], [line 1](vscode-notebook-cell:?execution_count=61&line=1)
----> [1](vscode-notebook-cell:?execution_count=61&line=1) epochs.save('example_data_epochs.fif', overwrite=True)

File <decorator-gen-222>:12, in save(self, fname, split_size, fmt, overwrite, split_naming, verbose)

File ~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:2325, in BaseEpochs.save(self, fname, split_size, fmt, overwrite, split_naming, verbose)
   [2322](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:2322)     # avoid missing event_ids in splits
   [2323](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:2323)     this_epochs.event_id = self.event_id
-> [2325](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:2325)     _save_split(this_epochs, split_fnames, part_idx, n_parts, fmt, overwrite)
   [2326](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:2326) return split_fnames

File ~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:136, in _save_split(epochs, split_fnames, part_idx, n_parts, fmt, overwrite)
    [133](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:133)     next_fname = split_fnames[next_idx]
    [135](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:135) with start_and_end_file(this_fname) as fid:
--> [136](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:136)     _save_part(fid, epochs, fmt, n_parts, next_fname, next_idx)

File ~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:149, in _save_part(fid, epochs, fmt, n_parts, next_fname, next_idx)
    [146](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:146)     write_id(fid, FIFF.FIFF_PARENT_BLOCK_ID, info["meas_id"])
    [148](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:148) # Write measurement info
--> [149](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:149) write_meas_info(fid, info)
    [151](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:151) # One or more evoked data sets
    [152](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:152) start_block(fid, FIFF.FIFFB_PROCESSED_DATA)
...
    [999](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/utils/numerics.py:999)         or dt.tzinfo is not timezone.utc
   [1000](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/utils/numerics.py:1000)     ):
-> [1001](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/utils/numerics.py:1001)         raise ValueError(f"Date must be datetime object in UTC: {repr(dt)}")

ValueError: Date must be datetime object in UTC: None
```

### Steps to reproduce

[... 22 lines cut ...]

fname_epochs = op.join(deriv_path, subject, 'ses-01', 'meg', subject + '_ses-01_task-audio_proc-clean_epo.fif')
epochs = mne.read_epochs(fname_epochs, preload=True)

epochs = epochs.anonymize()

epochs.save('example_data_epochs.fif', overwrite=True)
```

### Link to data

I can share the date privately, as I cannot anonymize them.

### Expected results

No errors.

### Actual results

Errors, see above.
The data is saved, but not complete.

### Additional information

Version: '1.9.0'

drammock, 2025-03-11:
I tried to repro locally and didn't reproduce (not a huge surprise) but I did run into a different problem:

1. load a `*proc-clean_epo.fif` from a MBP derivatives folder
2. call `epo.anonymize()`
3. call `epo.save()` -> `OverflowError: Python integer 2716099863 out of bounds for int32`

might make sense to fix both at the same time

larsoner, 2025-03-11:
Maybe we just need to change the check to:
```
if getattr(self, "annotations", None) is not None:
```

SophieHerbst, 2025-03-11:
that does fix the error with epochs.anonymize()
but I still cannot save the file, apparently a different problem: 

```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[7], [line 1](vscode-notebook-cell:?execution_count=7&line=1)
----> [1](vscode-notebook-cell:?execution_count=7&line=1) epochs.save('example_data_epochs.fif', overwrite=True)

File <decorator-gen-222>:12, in save(self, fname, split_size, fmt, overwrite, split_naming, verbose)

File ~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:2325, in BaseEpochs.save(self, fname, split_size, fmt, overwrite, split_naming, verbose)
   [2322](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:2322)     # avoid missing event_ids in splits
   [2323](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:2323)     this_epochs.event_id = self.event_id
-> [2325](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:2325)     _save_split(this_epochs, split_fnames, part_idx, n_parts, fmt, overwrite)
   [2326](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:2326) return split_fnames

File ~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:136, in _save_split(epochs, split_fnames, part_idx, n_parts, fmt, overwrite)
    [133](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:133)     next_fname = split_fnames[next_idx]
    [135](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:135) with start_and_end_file(this_fname) as fid:
--> [136](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:136)     _save_part(fid, epochs, fmt, n_parts, next_fname, next_idx)

File ~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:149, in _save_part(fid, epochs, fmt, n_parts, next_fname, next_idx)
    [146](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:146)     write_id(fid, FIFF.FIFF_PARENT_BLOCK_ID, info["meas_id"])
    [148](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:148) # Write measurement info
--> [149](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:149) write_meas_info(fid, info)
    [151](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:151) # One or more evoked data sets
    [152](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/epochs.py:152) start_block(fid, FIFF.FIFFB_PROCESSED_DATA)
...
    [999](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/utils/numerics.py:999)         or dt.tzinfo is not timezone.utc
   [1000](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/utils/numerics.py:1000)     ):
-> [1001](https://file+.vscode-resource.vscode-cdn.net/Users/sh254795/Documents/WORK_/TEACHING/2025_TEMPOMEGA_CAEN/~/.pyenv/versions/3.11.10/envs/mne/lib/python3.11/site-packages/mne/utils/numerics.py:1001)         raise ValueError(f"Date must be datetime object in UTC: {repr(dt)}")

ValueError: Date must be datetime object in UTC: None

```

SophieHerbst, 2025-03-11:
It seems to attempt to split the file so I tried downsampling and cropping to make it smaller, but I still run into the same error.

SophieHerbst, 2025-03-11:
```
epochs.info['meas_date']
datetime.datetime(2000, 1, 1, 0, 0, tzinfo=datetime.timezone.utc)
```

SophieHerbst, 2025-03-15:
I found the problem with saving the epochs file:

my info.get("helium_info") has no measurement date: 
<helium_info | he_level_raw: 49.15999984741211, meas_date: None>

But meas_info.py, l. 2883 asks to write it without checking whether it exists:
write_int(fid, FIFF.FIFF_MEAS_DATE, _dt_to_stamp(hi["meas_date"]))

Adding a line to check solves the error:
if hi.get("meas_date") is not None:
            write_int(fid, FIFF.FIFF_MEAS_DATE, _dt_to_stamp(hi["meas_date"]))

If you want, I can make a PR with this one and the change from above, but I don't oversee whether there is a bigger issue.

Also, these are the first recordings from our new MEG machine, should we lookin into the acquisition settings to make sure the helium measurement date is set?

larsoner, 2025-03-15:
Yes PR welcome!

Not sure about whether or not the system should have a meas date for the helium info. It's also possible it was never supposed to have it. But in any case I think MNE should preserve whatever is in the file so checking to see if `meas_date` is there and writing it if it is makes sense.

larsoner, 2025-07-07:
@SophieHerbst I looked into this briefly and I think this is probably already fixed in `main`, but I tacked on a little test to https://github.com/mne-tools/mne-python/pull/13310 and helped ensure things should work by using `.get` as well

closed 2025-07-10 by larsoner