mne-tools/mne-python #13157: evoked.plot_joint API `exclude` parameter docstring is inconsistent with the code

scott-huberty, 2025-03-12:
### Description of the problem

The [API doc](https://mne.tools/1.8/generated/mne.Evoked.html#mne.Evoked.plot_joint) Shows that the default value for parameter `exclude` is `'bads'`, but in the description for that parameter, it says that the default is `None`, when In fact, `None` is not even a valid argument for this parameter:

>     exclude : None | list of str | 'bads'
>        Channels names to exclude from being shown. If ``'bads'``, the
>       bad channels are excluded. Defaults to ``None``.

### Steps to reproduce

```Python
sample_fpath = mne.datasets.sample.data_path()
fpath = sample_fpath / "MEG" / "sample" / "sample_audvis_raw.fif"
raw = mne.io.read_raw(fpath, preload=True).pick("eeg")

epochs = mne.make_fixed_length_epochs(raw, duration=1, preload=True)
evoked = epochs.average()
# evoked.plot_joint(exclude=None) # include/exclude must be list, tuple, ndarray, or "bads". You provided type <class 'NoneType'>.
```

### Expected results

Figure shows with no bad channels excluded.

### Actual results

ValueError: include/exclude must be list, tuple, ndarray, or "bads". You provided type <class 'NoneType'>.

### Additional information

Should the code be fixed to match the docstring (i.e. accept `None`, and make it the default), or should the mention of `None` simply by removed from the docstring?

larsoner, 2025-03-13:
> Should the code be fixed to match the docstring (i.e. accept None, and make it the default), or should the mention of None simply by removed from the docstring?

Let's update the docstring to match the behavior

tharu-jwd, 2025-08-22:
Hi @scott-huberty @larsoner 

I've opened PR #13391 to fix this docstring inconsistency. The PR updates the docstring to match the actual behavior. Please review.

Thanks

PR #13391 (2025-08-22): MRG, DOC: Fix plot_evoked_joint exclude parameter default inconsistency
## Summary
Fixes inconsistency between the plot_evoked_joint function signature and its docstring regarding the exclude parameter default value.

## Problem
- evoked.plot_joint() method uses exclude='bads' as default
- plot_evoked_joint() function had exclude=None as default but docstring claimed it defaults to None
- This created API documentation inconsistency

## Solution
- Changed plot_evoked_joint default parameter from exclude=None to exclude='bads'
- Updated docstring to reflect the new default

## Testing
- Verifie [... cut]
closed 2025-08-22 by drammock