mne-tools/mne-python #13167: Incorrect datetime format parsing in NIRX file processing

ElblogdeIsmael, 2025-03-19:
### Description of the problem

---

### **Description of the problem**  
The function responsible for extracting the measurement date from NIRX files seems to contain an issue when parsing datetime formats. Specifically, the format list contains incorrectly formatted strings, which might lead to `ValueError` exceptions when `strptime` is called.  

One particular problem is the use of double quotes within the format strings:  

```python
for dt_code in [
    '"%a, %b %d, %Y""%H:%M:%S.%f"',
    '"%a %d %b %Y""%H:%M:%S.%f"',
    '"%a, %d %b %Y""%H:%M:%S.%f"',
    "%Y-%m-%d %H:%M:%S.%f",
    '"%Y年%m月%d日""%H:%M:%S.%f"',
]:
```
It looks like the intended format should not have double quotes around the datetime components.

---

### Steps to reproduce

```Python
# cd MNE PATH: MNE/mne-python/mne

pytest

### **Steps to reproduce**  

Attempting to process an NIRX file with a datetime format that does not match any of the valid formats will trigger a fallback to `January 1st, 2000`.  

Since `strptime` does not support the quoted format, the datetime parsing will silently fail.  

---
```

### Expected results

---

### **Expected results**  
The measurement date should be extracted correctly from the NIRX file without falling back to a default value.  

---

### Actual results

### **Actual results**  
The function fails to parse the datetime correctly, leading to the following warning:  

```
Extraction of measurement date from NIRX file failed.
This can be caused by files saved in certain locales (currently only [...] supported).
Please report this as a GitHub issue.
The date is being set to January 1st, 2000, instead of 'actual datetime string'.
```

### Additional information

---

### **Additional information**  
Should the datetime format strings be corrected to remove the unnecessary quotes? Alternatively, should the function handle errors more explicitly rather than relying on silent `ValueError` exceptions?  

---

drammock, 2025-03-19:
The strptime format strings found at 

https://github.com/mne-tools/mne-python/blob/1fb9dd6d5a3aa6734b9b07474c484d743338e8c2/mne/io/nirx/nirx.py#L240-L246

are based on *real, observed* timestamps in nirx files recorded by our users around the world. The presence of double-quotes in those format strings is intentional and necessary.

Side note:

> This issue was submitted on GitHub under the account ElBlogdeIsmael, though I primarily use
ismael-sallami.

it is a violation of GitHub's terms of use to have multiple (non-Pro) accounts.

closed 2025-03-19 by drammock

Ismael-Sallami, 2025-03-19:
Sorry for the inconvenience, it won't happen again and thanks for the reply.