mne-tools/mne-python #13168: Simulation of multivariate activation patterns based on experimental effects

AlexLepauvre, 2025-03-19:
### Describe the new feature or enhancement

I would like to propose adding a time-resolved multivariate simulation function to MNE-Python, based on the approach implemented in [SPM’s DEMO_CVA_RSA.m](https://github.com/spm/spm/blob/main/toolbox/DEM/DEMO_CVA_RSA.m). This method provides a simple framework for simulating structured multivariate activity patterns.

This feature extends their methodology by introducing time-resolved effects  to simulate dynamic experimental manipulations in EEG/MEG data. The simulated data would be returned as mne.Epochs objects, making it seamlessly compatible with existing MNE analysis pipelines.

Simulated EEG/MEG data is useful for:
- Validating multivariate decoding and RSA methods.
- Evaluating statistical robustness of EEG/MEG preprocessing techniques.
- ....

### Describe your proposed implementation

I have created a separate repository with a basic [implementation](https://alexlepauvre.github.io/meeg_simulator/api/meeg_simulator.html#submodules) with a [tutorial notebook](https://alexlepauvre.github.io/meeg_simulator/tutorial/using_meeg_simulator.html) showcasing its use. It's a simple function taking an experimental design, a noise parameter, number of channels, number of subjects, the effects that are supposed to be present as well as the time point at which those are supposed to be present

### Describe possible alternatives

I am currently considering the following improvements to what is currently there:
- Estimate spatial covariance from mne sample data sets
- Derive ERP from real data to make the simulated time resolved responses more reasonable (simple additive ERPs for each channel on top of the effects defined through FIR)
- Allow users to specify any basis function they see fit, rather than hard coding FIR within trial design matrix

### Additional context

This was done in collaboration with @qian-chu and we intend to write a small methods paper in JOSS to document the function.

We are unsure whether this should be incorporated in mne or be a separate toolbox. In the current state, the code is quite light, but we might consider adding features and expanding the code base:
- Simulate effects in specific frequency bands
- Forward modelling
- Simulate recording noise and artifacts (line noise, blinks...) to test entire preprocessing pipelines

qian-chu, 2025-03-19:
Thanks Alex! I just want to add that if the community see benefits in it being a MNE function, we can maybe initiate a PR to add the function to mne/simulation

larsoner, 2025-03-20:
See also https://github.com/mne-tools/mne-python/issues/12980. I think the class of possible things to simulate is quite large... maybe now is a good time to split off so things like this can be implemented more quickly!