mne-tools/mne-python #13170: Raw export edf file failed

BEGINRX, 2025-03-20:
### Description of the problem

I tried to export the data to edf format file. It failed.
The code is : 
`raw = mne.io.read_raw_edf(r"\\NITD-NAS\bci-rw\sleep\sleep-edf-database-expanded-1.0.0\sleep-edf-database-expanded-1.0.0\sleep-cassette\SC4001E0-PSG.edf")
raw.export(fname=r"E:\work\NITD\software_test\outputedf.edf", fmt="edf", physical_range="channelwise")`

And the error is 

[image]

### Steps to reproduce

```Python
raw = mne.io.read_raw_edf(r"\\NITD-NAS\bci-rw\sleep\sleep-edf-database-expanded-1.0.0\sleep-edf-database-expanded-1.0.0\sleep-cassette\SC4001E0-PSG.edf")
raw.export(fname=r"E:\work\NITD\software_test\outputedf.edf", fmt="edf", physical_range="channelwise")
```

### Expected results

Get the edf file.

### Actual results

Get the edf file.

### Additional information

Error.

Saksham8540, 2025-03-21:
Hi @BEGINRX ,  
I would like to work on this issue. Have you tried any other export formats, or does the error only happen with `fmt="edf"`? Also, could you share the exact error message to understand the problem better?  

Thanks!

BEGINRX, 2025-03-21:
Hi. I have tried other format like "brainvision" and "eeglab". The file were exported successfully. 

The error messages are only the ones listed above. There is no more information.

The file is in the attachment.

从网易163邮箱发来的超大附件推荐客户端极速下载
SC4001E0-PSG.edf (46.1M, 2025年4月5日 15:05 到期)
下载

mistraube, 2025-10-26:
To me this looks like a bug in the  [_round_float_to_8_characters](https://github.com/the-siesta-group/edfio/blob/main/edfio/edf_signal.py#L40) method from edfio. For `-2006585738.63211`, which is the value of pmin in the provided example data, it returns `-2006590000.0`.

Script to reproduce:
```python
#!/usr/bin/env python3

import mne
import math
import numpy as np
from typing import Callable

def _round_float_to_8_characters(
    value: float,
    round_func: Callable[[float], int],
) -> float:
    if isinstance(value, int) or value.is_integer():
        return value
    length = 8
    integer_part_length = str(value).find(".")
    if integer_part_length == length:
        return round_func(value)
    factor = 10 ** (length - 1 - integer_part_length)
    return round_func(value * factor) / factor

# https://physionet.org/files/sleep-edfx/1.0.0/sleep-cassette/SC4001E0-PSG.edf?download
raw = mne.io.read_raw_edf('SC4001E0-PSG.edf')

# relevant code from: _export_raw_edf_bdf()  
units = dict(
    eeg="uV", ecog="uV", seeg="uV", eog="uV", ecg="uV", emg="uV", bio="uV", dbs="uV"
)

ch_types = np.array(raw.get_channel_types())

ch_types_phys_max = dict()
ch_types_phys_min = dict()

for _type in np.unique(ch_types):
    _picks = [n for n, t in zip(raw.ch_names, ch_types) if t == _type]
    _data = raw.get_data(units=units, picks=_picks)
    ch_types_phys_max[_type] = _data.max()
    ch_types_phys_min[_type] = _data.min()

ch_type = ch_types[0]

pmin = ch_types_phys_min[ch_type]
pmax = ch_types_phys_max[ch_type]
if pmax == pmin:
    pmax = pmin + 1
prange = pmin, pmax

print("before _round_float_to_8_characters:", pmin)

# functions that are called by EdfSignal constructor
x = _round_float_to_8_characters(pmin, math.floor)
if float(x).is_integer():
    x = int(x)

print("after _round_float_to_8_characters:", x)

# trigger the error
raw.export("output.edf")
```