mne-tools/mne-python #13175: [ENH] Add uv environment support

alexrockhill, 2025-03-26:
### Describe the new feature or enhancement

It would be nice to have `uv` package manager support

### Describe your proposed implementation

Add `uv` install instructions to the documentation and update the pyproject.toml as needed to be `uv` compatible.

### Describe possible alternatives

n/a

### Additional context

`uv` is rust-based and much faster even than `mamba`. It could be that this is relatively easy to implement and might make for a faster install.

mscheltienne, 2025-03-30:
I don't think we need to edit anything in the `pyproject.toml`, I've been using `uv` for the past months.

alexrockhill, 2025-03-30:
Nice, I can just add a section in the docs.