mne-tools/mne-python #13191: Some limitations of EpochsTFRs: creating via raws, concatenating

mcfreund, 2025-04-05:
### Describe the new feature or enhancement

My pipeline involves cropping raws into blocks of trials, time-frequency transforming these cropped raws, then epoching them into trials. Transforming raws rather than epochs is desirable as it avoids creating edge artifacts around every trial. Under this workflow, it would be convenient if EpochsTFRArray could accept RawTFR objects and epoch them, as Epochs can do with Raws.

Another limitation of working with EpochsTFRs is concatenation. I've constructed EpochsTFRs from multiple different blocks/sessions and would like to concatenate them. I had expected concatenate_epochs to accept EpochsTFR but unfortunately it does not. see also https://mne.discourse.group/t/concatenating-epochstfr-objects/8463

### Describe your proposed implementation

Either modification of `concatenate_epochs` and `EpochsTFRArray`, or new functions/classes analogous to time-domain counterparts.

### Describe possible alternatives

As a work-around, I am flattening the time-frequency data arrays, concatenating channels and frequencies, then working with them as Epochs objects (epoching, concatenating). But the Epochs class is ill suited for handling time-frequency data.

drammock, 2025-04-06:
> it would be convenient if EpochsTFRArray could accept RawTFR objects and epoch them, as Epochs can do with Raws.

For parity with the time domain classes, shouldn't it be `EpochsTFR` (not `EpochsTFRArray`) that did that? 

> I had expected concatenate_epochs to accept EpochsTFR but unfortunately it does not.

I agree that it should. Are you up for making a PR to add this? 

I'd guess that concatenation will be easier than the other change, and I'd suggest tackling these in separate PRs

drammock, 2025-04-10:
I noticed yesterday on a dataset that I'm working on that `concatenate_epochs` also does not work with `EpochsSpectrum` objects. If it's not too much extra work, would be great if support for that were added at the same time as `EpochsTFR` support.

drammock, 2026-03-12:
@aman-coder03 there are 3 tasks here:

1. make `concatenate_epochs` work for `EpochsTFR` objects.
2. make `concatenate_epochs` work for `EpochsSpectrum` objects.
3. make `EpochsTFR` accept `RawTFR` objects as input.

tasks 1 and 2 could be done in a single PR or separately. Task 3 is conceptually distinct from the other two and should be its own PR (and will probably be harder)