mne-tools/mne-python #13193: test failures with scipy 1.15.0 - sph_harm deprecation

emollier, 2025-04-06:
### Description of the problem

When running the MNE Python 1.9.0 test suite in Debian sid, we are running into [Debian bug #1100273](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1100273).

### Steps to reproduce

```Python
python3 -m pytest mne/preprocessing/tests/test_maxwell.py
python3 -m pytest mne/tests/test_transforms.py
```

### Expected results

I would expect the affected test items to pass.  :)

### Actual results

Test items fail with the following tracebacks:
```
mne/preprocessing/tests/test_maxwell.py:434: in test_spherical_conversions
    sph = sph_harm(order, degree, az, pol)
E   SystemError: <class 'DeprecationWarning'> returned a result with an exception set
```
```
mne/tests/test_transforms.py:87: in test_tps
    warp.fit(source[::3], destination[::2])
<decorator-gen-50>:12: in fit
    ???
mne/transforms.py:1164: in fit
    src_sph = _compute_sph_harm(order, *src_rad_az_pol[1:])
mne/transforms.py:929: in _compute_sph_harm
    sph = sph_harm(order_, degree, az, pol)
E   SystemError: <class 'DeprecationWarning'> returned a result with an exception set
```

### Additional information

Issues look related to [scipy.special.sph_harm](https://docs.scipy.org/doc/scipy-1.15.0/reference/generated/scipy.special.sph_harm.html#scipy.special.sph_harm) deprecation in version 1.15.0.  I have attempted to see what happens when moving to the replacement [sph_harm_y](https://docs.scipy.org/doc/scipy-1.15.0/reference/generated/scipy.special.sph_harm_y.html#scipy.special.sph_harm_y), but it seems to calculate harmonics slightly differently:
```
mne/preprocessing/tests/test_maxwell.py:436: in test_spherical_conversions
    assert_allclose(_sh_negate(sph, order), sph_harm_y(-order, degree, az, pol))
E   AssertionError: 
E   Not equal to tolerance rtol=1e-07, atol=0
E   
E   Mismatched elements: 580 / 600 (96.7%)
E   Max absolute difference among violations: 0.34498745
E   Max relative difference among violations: inf
E    ACTUAL: array([[ 0.000000e+00+0.000000e+00j,  7.427103e-02+0.000000e+00j,
E            1.450692e-01+0.000000e+00j,  2.090842e-01+0.000000e+00j,
E            2.633225e-01+0.000000e+00j,  3.052482e-01+0.000000e+00j,...
E    DESIRED: array([[0.+0.j, 0.+0.j, 0.+0.j, 0.+0.j, 0.+0.j, 0.+0.j, 0.+0.j, 0.+0.j,
E           0.+0.j, 0.+0.j, 0.+0.j, 0.+0.j, 0.+0.j, 0.+0.j, 0.+0.j, 0.+0.j,
E           0.+0.j, 0.+0.j, 0.+0.j, 0.+0.j, 0.+0.j, 0.+0.j, 0.+0.j, 0.+0.j,...
```
```
mne/tests/test_transforms.py:90: in test_tps
    assert_allclose(destination_est, destination, atol=1e-3)
E   AssertionError: 
E   Not equal to tolerance rtol=1e-07, atol=0.001
E   
E   Mismatched elements: 42 / 600 (7%)
E   Max absolute difference among violations: 0.00437203
E   Max relative difference among violations: 0.00591184
E    ACTUAL: array([[ 1.564246e+00,  6.922982e-06,  1.918821e+00],
E          [ 2.081207e+00, -2.403024e-06,  1.682256e+00],
E          [ 2.512893e+00,  1.372459e-05,  1.308731e+00],...
E    DESIRED: array([[ 1.563465e+00,  0.000000e+00,  1.918986e+00],
E          [ 2.081282e+00,  0.000000e+00,  1.682507e+00],
E          [ 2.511499e+00,  0.000000e+00,  1.309721e+00],...
```

larsoner, 2025-04-06:
Could it be "just" a sign difference that we need to account for / add?

larsoner, 2025-04-09:
This has already been fixed in `main` in https://github.com/mne-tools/mne-python/pull/13049

closed 2025-04-09 by larsoner

emollier, 2025-04-09:
Perfect!  We'll be able to drop our workaround on next release.

Thank you  :)