mne-tools/mne-python #13194: Proposed Tutorial: Creating MNE-Python Data Structures (Info, Raw, Epochs, Evoked) from Scratch

Ganasekhar-gif, 2025-04-07:
### Proposed documentation enhancement

I would like to propose adding a new tutorial to the simulation section of the MNE-Python documentation that demonstrates how to manually construct the core MNE data structures — Info, Raw, Epochs, and Evoked — from scratch using simulated or dummy data.

Why this is useful
While MNE-Python provides powerful tools for working with real MEG/EEG data, newcomers and educators often need a clear, step-by-step demonstration of how its core data structures are built internally. This tutorial would:
Help contributors and developers gain a deeper understanding of MNE internals.
Assist educators and students in learning MNE without needing external data files.
Serve as a foundation for unit testing, simulation-based analysis, and benchmarking.
Be useful for those working on projects where synthetic or simulated data is required.

Proposed File and Location
tutorials/simulation/plot_create_mne_objects_from_scratch.py
This file will follow the existing tutorial format and live alongside other simulation-focused tutorials.

Structure of the Tutorial
The tutorial will include:
Creating an Info object – with basic channel metadata
Creating dummy data and building a RawArray
Defining synthetic events and constructing Epochs
Averaging epochs to form an Evoked object
Visualizing all the structures

The tutorial will follow MNE’s Sphinx-Gallery format, including inline plotting, explanations, and proper docstring headers.

Dependencies
Standard mne and numpy only
No external datasets required

Suggested Labels
documentation, enhancement, tutorial

Commitment
I am committed to fully implementing this tutorial and responding to all review feedback. This issue is also aligned with my GSoC proposal to contribute educational content and documentation to MNE-Python.

Prasadmaganti, 2025-04-07:
Hi, I’d like to work on this tutorial as part of my contributions to MNE-Python. I’ll follow the Sphinx-Gallery format to demonstrate how to create Info, Raw, Epochs, and Evoked objects using simulated data with only numpy and mne.

drammock, 2025-04-07:
That tutorial *already exists* in the simulation section: https://mne.tools/dev/auto_tutorials/simulation/10_array_objs.html

closed 2025-04-07 by drammock

Ganasekhar-gif, 2025-04-08:
Hi drammock,

Thanks for the clarification and for pointing to the existing tutorial at 10_array_objs.py — I appreciate it!

I had intended to create a more beginner-focused version with detailed step-by-step explanations, extra visualizations (like ERP plots, topomaps at peak latencies), and narrative comments to help newcomers understand the flow from Info → Raw → Epochs → Evoked.

Would it be okay if I worked on enhancing the existing tutorial with additional comments and visuals for better pedagogy? Or would you prefer I propose a follow-up tutorial that goes one step further — perhaps showing how to simulate event-related potentials and analyze them with time-frequency techniques?

Let me know how I can best contribute — happy to adapt and help where needed!

Thanks again

drammock, 2025-04-08:
> I had intended to create a more beginner-focused version with detailed step-by-step explanations, extra visualizations (like ERP plots, topomaps at peak latencies), and narrative comments to help newcomers understand the flow from Info → Raw → Epochs → Evoked.

Well, we also already have tutorials that cover that information. [The very first tutorial](https://mne.tools/dev/auto_tutorials/intro/10_overview.html#sphx-glr-auto-tutorials-intro-10-overview-py) covers Raw → Epochs → Evoked; plotting spectra, time-frequency, ERP/ERF, topomaps; and also inverse modeling.

> Would it be okay if I worked on enhancing the existing tutorial with additional comments and visuals for better pedagogy? Or would you prefer I propose a follow-up tutorial that goes one step [...]

I would be really happy if you identified parts of our tutorials that were confusing and suggested ways to make them easier to understand. I'd also be happy to add new tutorials *for topics that we don't already cover*, or for topics that so far are only covered in the "examples" format (which are meant to be quick how-to recipes rather than detailed pedagogical material).

What I'd like to avoid is you creating a new tutorial that combines bits and pieces from other tutorials that already exist, because we're not likely to add tutorials that are redundant with what we already have.  Also, I'm not really in favor of adding really long tutorials that try to show everything a user might want all on one page. IMO it makes the information harder to find. Our very first tutorial (mentioned above) is an exception to that, as it's designed to be a super-high-level overview (but there, for example, the time-frequency section is just 3 lines of code and 1 plot; there are 3 entire separate tutorials that cover time-frequency in much greater detail).