mne-tools/mne-python #13204: Showing label borders is not possible on the flat brain surface

ctrltz, 2025-04-11:
### Description of the problem

Hi and thank you for all the effort put into MNE-Python! Unfortunately, I ran into a problem when trying to plot borders of any region of the Desikan-Killiany parcellation on the flat brain surface. Somewhat surprisingly to me, plotting the parcellation as a whole works fine, although it also requires showing borders of all regions. Plotting the whole region (`borders=False`) also works, the problem only occurs with `borders=True` or borders set to an integer.

I would be happy to fix the issue but it might take a while for me to figure out the cause. If you have any hints about what went wrong, they would help a lot :)

### Steps to reproduce

```Python
import mne

from mne.datasets import sample 

subjects_dir = sample.data_path() / "subjects"
subject = "fsaverage"

labels = mne.read_labels_from_annot(subject, "aparc", subjects_dir=subjects_dir)

Brain = mne.viz.get_brain_class()
brain = Brain(subject=subject, hemi="lh", surf="flat")
# brain.add_annotation("aparc")               # works fine
# brain.add_label(labels[0], borders=False)   # works fine
brain.add_label(labels[0], borders=True)      # crashes
brain._renderer.plotter.app.exec()
```

### Expected results

A Pyvista figure of the flat surface of the left hemisphere with borders of one label (bankssts-lh) shown

### Actual results

Error message:

```
Traceback (most recent call last):
  File "/home/willenjoy/Projects/roiextract-simulations/scripts/misc/bugs/mne_viz_brain_add_label_flat.py", line 14, in <module>
    brain.add_label(labels[0], borders=True)      # crashes
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/willenjoy/miniforge3/envs/roiextract-simulations/lib/python3.12/site-packages/mne/viz/_brain/_brain.py", line 2265, in add_label
    keep_idx = _mesh_borders(self.geo[hemi].faces, scalars)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/willenjoy/miniforge3/envs/roiextract-simulations/lib/python3.12/site-packages/mne/surface.py", line 1929, in _mesh_borders
    edges = mesh_edges(tris)
            ^^^^^^^^^^^^^^^^
  File "/home/willenjoy/miniforge3/envs/roiextract-simulations/lib/python3.12/site-packages/mne/surface.py", line 1713, in mesh_edges
    return _mesh_edges(tris=tris)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/willenjoy/miniforge3/envs/roiextract-simulations/lib/python3.12/site-packages/mne/surface.py", line 1719, in _mesh_edges
    raise ValueError("Cannot compute adjacency on a selection of triangles.")
ValueError: Cannot compute adjacency on a selection of triangles.
```

### Additional information

```
Platform             Linux-6.8.0-57-generic-x86_64-with-glibc2.35
Python               3.12.0 | packaged by conda-forge | (main, Oct  3 2023, 08:43:22) [GCC 12.3.0]
Executable           /home/willenjoy/miniforge3/envs/roiextract-simulations/bin/python
CPU                  12th Gen Intel(R) Core(TM) i7-1255U (12 cores)
Memory               15.3 GiB

Core
├☑ mne               1.9.0 (latest release)
├☑ numpy             1.26.2 (OpenBLAS 0.3.23.dev with 12 threads)
├☑ scipy             1.14.1
└☑ matplotlib        3.8.2 (backend=QtAgg)

Numerical (optional)
├☑ sklearn           1.3.2
├☑ numba             0.61.0
├☑ nibabel           5.2.0
├☑ pandas            2.1.4
└☐ unavailable       nilearn, dipy, openmeeg, cupy, h5io, h5py

Visualization (optional)
├☑ pyvista           0.43.1 (OpenGL 4.6 (Core Profile) Mesa 23.2.1-1ubuntu3.1~22.04.2 via Mesa Intel(R) Graphics (ADL GT2))
├☑ pyvistaqt         0.11.0
├☑ vtk               9.3.0
├☑ qtpy              2.4.1 (PyQt6=6.8.1)
└☐ unavailable       ipympl, pyqtgraph, mne-qt-browser, ipywidgets, trame_client, trame_server, trame_vtk, trame_vuetify

Ecosystem (optional)
├☑ mne-bids          0.16.0
├☑ mne-connectivity  0.7.0
└☐ unavailable       mne-nirs, mne-features, mne-icalabel, mne-bids-pipeline, neo, eeglabio, edfio, mffpy, pybv
```

scrharsh, 2025-04-18:
I've opened this PR and done my best to implement the changes as discussed. When you get a chance, I’d really appreciate it if you could take a look and share your feedback. Looking forward to your review. Thanks.

ctrltz, 2026-10-20:
Found this old issue that I reported - it turns out that it was fixed in #14308, and I can't reproduce it on the current `main` branch. Closing.

closed 2026-10-20 by ctrltz

larsoner, 2026-10-22:
Great, thanks for following up @ctrltz !