mne-tools/mne-python #13216: Add a "neighbor" method to `interpolate_bads` that works without exact sensor positions

sappelhoff, 2025-04-16:
### Describe the new feature or enhancement

[`interpolate_bads`](https://mne.tools/stable/generated/mne.io.Raw.html#mne.io.Raw.interpolate_bads) has several methods implemented, like the most commonly used spherical spline approach, but also "mne", and "nearest" (for  fnirs, although it is not documented what this entails).

All of these methods require 3D sensor positions to work, however, sometimes these may not be available.

As an alternative, it would be great to pass an adjacency matrix (see e.g., [`find_ch_adjacency`](https://mne.tools/stable/generated/mne.channels.find_ch_adjacency.html)), and have the interpolation be done as a (weighted) average of the neighbors.

This is something that is already supported in FieldTrip: https://github.com/fieldtrip/fieldtrip/blob/55e8f26facac8eaf16ba7222433eba3080803b5c/ft_channelrepair.m#L3-L7

### Describe your proposed implementation

1. There should be a function to easily create an "mne adjacency matrix" from scratch, for example by passing a dict of neighbors:

```python
mydict = {
    ch1: [ch2, ch3],  # mapping a channel to its neighbors
    ...
}
adjacency = mne.channels.make_adjacency(mydict)
```

2. `.interpolate_bads` method should have a new `method` argument called `neighbors`

### Describe possible alternatives

I am not aware of an alternative for channel interpolation when no sensor positions are available.

larsoner, 2025-04-17:
I think idea (2) is fine.

For idea (1) I think I'd prefer to help people understand a bit how to interpret and construct an adjacency matrix. At the end of the day it's an ndarray of shape `(n_ch, n_ch)` and you put 1's where channels are adjacent. So the hard part is really constructing a suitable `mydict`. Once you have that the helper you're talking about is very short and I think not too hard to understand:
```
adj = np.zeros((len(ch_names), len(ch_names))
for fro, tos in mydict.items():
    ii = ch_names.index(fro)
    for to in tos:
        jj = ch_names.index(to)
        adj[ii, jj] = adj[jj, ii] = 1
```
So is probably better suited to an example/tutorial that talks about what an adjacency matrix is.

drammock, 2025-04-17:
+1 for option (2).  I prefer the name `"custom"` rather than `"neighbors"` for the method where users pass in a custom adjacency matrix.