mne-tools/mne-python #13227: ENH: Add support for tracking arbitrary sensor transformation matrices

larsoner, 2025-04-24:
In FieldTrip / SPM there is a notion of a `tra` matrix, which is "whatever transformation has been applied to the sensor space data" (could be ICA, SSP/projections, reference sensor synthetic gradiometry, or really any arbitrary transformation). It is important to keep track of this transformation because during source imaging the forward matrix needs to have the same transformation applied. In MNE we currently have:

1. `comp` matrices, which are usually used for CTF systems and have to do with gradient compensation
2. `proj` matrices, which are used for SSP vectors and set a null space

So this enhancement proposal is to add support for `transformation` matrices, which are arbitrary arrays of shape `(n_ch, n_ch)`. Implementation-wise, it would require:

1. adding a new FIF constant for this matrix
2. adding an interface for adding them, e.g., `raw.apply_transformation(transformation, picks=None)` or `raw.add_transformation` (advantages/disadvantages to both we can discuss later) that applies it (probably no option to delay like projectors in the first iteration) and attaches it as a new `info["transformation"]` or similar
3. modifying our channel-picking code to prohibit picking channels when such a transformation has been applied and picking would lead to losing mixed channels (e.g., if your transformation was a fully-nonzero matrix but just for MEG channels, you couldn't `.pick(["MEG 0111"])` but you could `.pick("eeg")` or `.pick("meg")`)
4. modifying our forward sol prep code to support applying it just like we do for `info["comp"]` and `info["projs"]`

A few good motivations for this:

1. Facilitate interoperability with FT/SPM
2. Facilitate use with other preprocessing techniques
3. Better accounting of our own preprocessing. In particular I'm thinking that `ica.apply` should set this value. Currently we assume that during source imaging our pre-whitening step takes care of any distortions caused by `ica.apply(inst)`, but this would help ensure it.

@agramfort any thoughts on this, having worked with ICA and source imaging and also understanding a bit our fwd/inv code? I don't forsee this adding too much complexity but maybe I'm missing something.

cc @britta-wstnr @schoffelen in case I (inevitably) missed something from our discussion

<details>
<summary>Some other details discussed that probably make sense</summary>

We could start by allowing `transformation` to be any matrix (easiest). Then we could look at whether it's actually a projection, and add it as `proj`; or if it is rank-deficient, decomposing into a projection part and a non-projection-part via SVD, which would have the advantage of working nicely with our existing `proj` code.

Simple unit tests could be written to start by having `transformation` just be a permutation of sensors (i.e., `transformation = np.eye(n_ch)[rng.permutation(n_ch)]` should give identical imaging results as unpermuted data.

</details>

schoffelen, 2025-04-24:
Hi @larsoner I think this is a good summary. Just for clarification and reference, and to avoid semantic confusion later on:

What in FT is known as a 'tra' matrix is a (integration-point-of-a-magnetometer-)coil-to-channel mapping matrix that typically accumulates - throughout the course of an analysis pipeline - a sequence of projections (e.g. mapping into and out of an IC-space, addition of synthetic gradients etc). In the sensor array object we keep track of the order of the individual balancing steps (along with the corresponding nchan x nchan projection matrices). Up to a certain level this allows for moving back and forth (as long as no information is lost along the way). The channel-level data always passes through the individual linear transformation steps along the way. In FieldTrip the generic projections are also known as 'montages' (equivalent to how this is used in EEG), and https://github.com/fieldtrip/fieldtrip/blob/master/forward/ft_apply_montage.m is the function that does the bookkeeping and applies the projection to the relevant numeric data.
In high level functions that apply transformations, e.g. ft_denoise_synthetic, which does CTF 3d-order gradient compensation, the projections are applied to both data and sensor array, starting from https://github.com/fieldtrip/fieldtrip/blob/603f3050c8e35750311066b18200a0551ec5ea84/ft_denoise_synthetic.m#L123

larsoner, 2025-08-22:
As a start, we should probably modify the FT reader code. If the `tra` matrix is present and it's not an identity matrix, we should warn people that source imaging (and maybe other steps since it could even be a channel remapping?) might be compromised and to proceed with caution. That would give us some sense of the impact of this on end users and also maybe prevent some mistakes until we fully support `tra`