mne-tools/mne-python #13230: save(): Error message is unclear

sjentsch, 2025-04-27:
https://github.com/mne-tools/mne-python/blob/8f83332cac8b0cdbc28ca038ef4a66c6f876dc9c/mne/io/base.py#L1795 tests for both, whether the data have been preloaded and `fname`. `fname` again is constructed some lines above, considering what is set in the parameter `overwrite`.

One would expect that `overwrite` is the only parameter that counts, and wonder why one still gets an error message. The error message should be either split, saying sth. like “When overwriting files, data need to be preloaded” or the current error message should add such information. Without this information, the user is puzzled (given that `overwrite` was set to `True`).

I suppose, the same applies to `save` in Epochs (and perhaps in other classes too).

Thanks!

agramfort, 2025-04-27:
in what context do you bug into a weird error message? The error you refer
to only happens if preload is True.
in this case the data in not in memory but still on disk so if you try to
save to the same file with overwrite
you will delete the source data you can to copy. Does that make sense?

> Message ID: ***@***.***>
>

sjentsch, 2025-04-27:
No, my point is that I set the parameter `overwrite = True`. The Value Error that is thrown ("You cannot save data to the same file. Please use a different filename.") is puzzling because I set `overwrite = True`. It is not obvious `overwrite = True` only works if one also sets `preload = True`. My thinking was, I set `overwrite = True`, what do I get the error message. Only by looking into the code, I could figure out that the data also have to be preloaded for `overwrite = True` to work as expected (overwrite the file).

I completely understand why the data have to be preloaded, but it would be helpful if there is an error message informing one about that, like:

```
        if not self.preload and fname in self.filenames:
            if overwrite:
                raise ValueError(
                    "In order to save data to the same file, data need to be preloaded "
                    "AND overwrite needs to be set to True"
                )
            else:
                raise ValueError(
                    "You cannot save data to the same file. Please use a different "
                    "filename."
                )
```

larsoner, 2025-04-28:
Agreed it makes sense to improve the err message. But I would do it with something like
```
        if not self.preload and fname in self.filenames:
            extra = " and overwrite must be True" if not overwrite else ""
            raise ValueError(
                "In order to save data to the same file, data need to be preloaded "
                + extra)
```
to be a tiny bit more DRY. @sjentsch would you be up for making a PR to improve the message?

PR #13233 (2025-04-28): Improve error message for overwrite
#### Reference issue (if any)

Fixes #13230.

#### What does this implement/fix?

Provides an error message that makes more clear that data need to be preloaded if one wishes to overwrite files (`save()` with the parameter `overwrite=True`).
closed 2025-04-28 by larsoner