mne-tools/mne-python #13238: apply ICA method

MegaSear, 2025-05-03:
### Description of the problem

_apply_raw contains "picks = pick_types( raw.info, meg=False, include=self.ch_names, exclude="bads", ref_meg=False)" 
But ICA.fit contains "picks = _picks_to_idx(inst.info, picks, allow_empty=False, with_ref_meg=self.allow_ref_meg)"
Interesting, I can train ICA on raw with bads channels 
But using the same raw in the apply method I get an error that there are less channels due to bads

### Steps to reproduce

```Python
raw.info['bads'] = ['Fpz']
n_components = 16
ica = ICA(n_components=n_components)
ICA.fit(raw)
ICA.apply(raw, exclude=[])
```

### Expected results

The apply method is not expected to remove bad channels

### Actual results

"operands could not be broadcast together with shapes (54,2455179) (63,1) (54,2455179)".

### Additional information

Platform             Windows-10-10.0.19045-SP0
Python               3.10.8 (tags/v3.10.8:aaaf517, Oct 11 2022, 16:50:30) [MSC v.1933 64 bit (AMD64)]
Executable           C:\Users\searg\AppData\Local\Programs\Python\Python310\python.exe
CPU                  Intel(R) Core(TM) i5-9300H CPU @ 2.40GHz (8 cores)
Memory               15.8 GiB

Core
 + mne               1.9.0 (latest release)
 + numpy             1.26.4 (OpenBLAS 0.3.23.dev with 8 threads)
 + scipy             1.15.2
 + matplotlib        3.10.1 (backend=qtagg)

Numerical (optional)
 + sklearn           1.6.1
 + numba             0.59.1
 + cupy              12.3.0
 + pandas            2.2.3
 + h5io              0.2.4
 + h5py              3.13.0
 - unavailable       nibabel, nilearn, dipy, openmeeg

larsoner, 2025-05-06:
Any chance you could make a fully reproducible example that could be run using the `sample` dataset?

MegaSear, 2025-05-06:
import mne 
from mne.preprocessing import ICA

from mne.datasets import eegbci
from mne.io import read_raw_edf
from pathlib import Path
import os 

mne_data_dir = Path.home() / 'mne_data'
mne_data_dir.mkdir(parents=True, exist_ok=True)

os.environ['MNE_DATA'] = str(mne_data_dir)
mne.set_config('MNE_DATA', os.environ['MNE_DATA'], set_env=True)

subject = 1
runs = [6]
edf_paths = eegbci.load_data(subject, runs)
raw = read_raw_edf(edf_paths[0], preload=True)

print(raw.info)

raw.info["bads"] = [raw.ch_names[0]] # the first channel was chosen for nothing
raw.info["bads"]

example_channels = raw.ch_names

ica = ICA(n_components=20) # the 20 components was chosen for nothing

ica.fit(inst=raw.copy(), picks=example_channels) # pick ALL channels

print(ica.pca_components_.shape)
print(ica.mixing_matrix_.shape)
print(ica.get_sources(raw.copy()).get_data().shape)

ica.apply(inst=raw.copy(), exclude=[])