mne-tools/mne-python #13244: Pupil size unit

auracionjyl, 2025-05-12:
### Description of the problem

MNE version: 1.9.0
Operating system: Windows 11

When I was using the following code to read eye-movement data from .asc file. I found that the values of pupil size (i.e., pupil_right/pupil_left) were extremely large. They were typically 1000-5000 in v1.7.0, while they were 10^6 times larger in v1.9.0 (the raw data are the same). The issue still existed when I removed `mne.preprocessing.eyetracking.convert_units`.
```
cals = read_eyelink_calibration(self.edata_path)
first_cal = cals[0]
first_cal["screen_resolution"] = (2560, 1440)
first_cal["screen_size"] = (0.6, 0.35)
first_cal["screen_distance"] = 0.62
raw_edata = read_raw_eyelink(self.edata_path, create_annotations=["blinks", "saccades", "fixations", "messages"])
mne.preprocessing.eyetracking.convert_units(raw_edata, calibration=first_cal, to="radians")
raw_pick = raw_edata.pick(picks="eyetrack")
data = raw_pick.to_data_frame()
```

In v1.7.0, it is arbitrary units (AU), which makes sense. What is the pupil size unit in v1.9.0? I noticed that the unit is micrometers in the [figure](https://mne.tools/stable/auto_tutorials/preprocessing/90_eyetracking_data.html#plot-the-raw-eye-tracking-data) given in the document. However, the pupil size (pupil_right) is about 2x10^9 micrometers (i.e., 2km), which is impossible. Did I miss a parameter in any function? Or it is actually a bug.

Plus, I notice that it says 

> MNE-Python expects eyegaze positions to be in radians of visual angle, and pupil size to be in meters.

in the [document](https://mne.tools/stable/auto_tutorials/preprocessing/90_eyetracking_data.html#standardizing-eyetracking-data-to-si-units). It could be better if the pupil size is millimeters or micrometers (AU is OK as well). Meters may be too large to describe pupil size.

### Steps to reproduce

```Python
cals = read_eyelink_calibration(self.edata_path)
first_cal = cals[0]
first_cal["screen_resolution"] = (2560, 1440)
first_cal["screen_size"] = (0.6, 0.35)
first_cal["screen_distance"] = 0.62
raw_edata = read_raw_eyelink(self.edata_path, create_annotations=["blinks", "saccades", "fixations", "messages"])
mne.preprocessing.eyetracking.convert_units(raw_edata, calibration=first_cal, to="radians")
raw_pick = raw_edata.pick(picks="eyetrack")
data = raw_pick.to_data_frame()
```

### Link to data

I'm sorry that I can't upload an .asc file, but I believe that you could evaluate with any .asc file.

### Expected results

The values of pupil_right/pupil_left in `data` are typically 1000-5000. It is true in v1.7.0.

### Actual results

The values in v1.9.0 are exactly 10^6 times larger than in v1.7.0.

### Additional information

v1.7.0:
Platform             Windows-11-10.0.22631-SP0
Python               3.12.4 (tags/v3.12.4:8e8a4ba, Jun  6 2024, 19:30:16) [MSC v.1940 64 bit (AMD64)]
Executable           C:\Program Files\Python312\python.exe
CPU                  Intel64 Family 6 Model 167 Stepping 1, GenuineIntel (16 cores)
Memory               Unavailable (requires "psutil" package)
Core
├☒ mne               1.7.0 (outdated, release 1.9.0 is available!)
├☑ numpy             2.2.5 (unknown linalg bindings (threadpoolctl module not found: No module named 'threadpoolctl'))
├☑ scipy             1.15.3
└☑ matplotlib        3.10.3 (backend=module://backend_interagg)
Numerical (optional)
├☑ pandas            2.2.3
└☐ unavailable       sklearn, numba, nibabel, nilearn, dipy, openmeeg, cupy, h5io, h5py
Visualization (optional)
└☐ unavailable       pyvista, pyvistaqt, vtk, qtpy, ipympl, pyqtgraph, mne-qt-browser, ipywidgets, trame_client, trame_server, trame_vtk, trame_vuetify
Ecosystem (optional)
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline, neo, eeglabio, edfio, mffpy, pybv

v1.9.0:
Platform             Windows-11-10.0.22631-SP0
Python               3.12.4 (tags/v3.12.4:8e8a4ba, Jun  6 2024, 19:30:16) [MSC v.1940 64 bit (AMD64)]
Executable           C:\Program Files\Python312\python.exe
CPU                  11th Gen Intel(R) Core(TM) i9-11900K @ 3.50GHz (16 cores)
Memory               63.7 GiB
Core
 + mne               1.9.0 (latest release)
 + numpy             2.2.5 (unknown linalg bindings (threadpoolctl module not found: No module named 'threadpoolctl'))
 + scipy             1.15.3
 + matplotlib        3.10.3 (backend=module://backend_interagg)
Numerical (optional)
 + pandas            2.2.3
 - unavailable       sklearn, numba, nibabel, nilearn, dipy, openmeeg, cupy, h5io, h5py
Visualization (optional)
 - unavailable       pyvista, pyvistaqt, vtk, qtpy, ipympl, pyqtgraph, mne-qt-browser, ipywidgets, trame_client, trame_server, trame_vtk, trame_vuetify
Ecosystem (optional)
 - unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline, neo, eeglabio, edfio, mffpy, pybv

larsoner, 2025-07-03:
@drammock will you have time to look at this next week?

drammock, 2025-07-04:
internally, default (plotting/axis-label) unit for `pupil` is µm

https://github.com/mne-tools/mne-python/blob/6ad882335644fb80f8744e508387c52f7231553a/mne/defaults.py#L96

and accordingly the default scaling is 1e6:

https://github.com/mne-tools/mne-python/blob/6ad882335644fb80f8744e508387c52f7231553a/mne/defaults.py#L125

therefore `_handle_default("scalings", None)` returns a dict with entry `"pupil": 1000000.0`.

The default scaling for `pupil` channels was changed from 1 to 1e6 in #12846 by @scott-huberty.  I don't really know off-the-cuff what the right move is here; AFAICT the `to_data_frame` method is working as expected.

PR #13314 (2025-07-11): Adjust Pupil channel units (again)
closes #13244 

There have been several iterations on the issue of eyetracking units, so let me summarize for brevity:

Stemming from Issue https://github.com/mne-tools/mne-python/issues/12756#issuecomment-2260832644 (and PR https://github.com/mne-tools/mne-python/pull/12846#issuecomment-2349960065 + https://github.com/mne-tools/mne-python/pull/12850), we changed MNE internals to assume all incoming Eyetracking data were already represented in Meters (SI unit of distance, as _some_ eyetrackers report pupil size in mm). In those PR's I adjusted the scaling for [... cut]