mne-tools/mne-python #13257: info['file_id'] not set properly when there is a directory

papadop, 2025-05-20:
### Description of the problem

When a FIF file contains a directory, the `file_id` item of the info dictionary remains to `None`. This is because the _read_dir_entry_struct ignores (wrongly in my opinion) the first tag in the directory, which happens to be the file_id. Alternatively, the file_id has already been read in open.py (_fiff_open), but the id is not used.

Correction:
Change _read_dir_entry_struct to:
```
def _read_dir_entry_struct(fid, tag, shape, rlims):
    """Read dir entry struct tag."""
    pos = tag.pos
    entries = list()
    for offset in range(tag.size // 16):
    ....
```
where ... mean "code unchanged".

I can create a patch if the analysis above is validated...

### Steps to reproduce

```Python
import mne
sample_data_folder = mne.datasets.sample.data_path()
raw = mne.io.read_raw(sample_data_folder / "MEG" / "sample" / "sample_audvis_filt-0-40_raw.fif")
print(raw.info['file_id'])
```

### Expected results

{'version': 65539, 'machid': array([-1137681485,  1834156032], dtype='>i4'), 'secs': 1384803104, 'usecs': 923597}

### Actual results

None

### Additional information

Platform             Linux-6.14.6-300.fc42.x86_64-x86_64-with-glibc2.41
Python               3.13.3 (main, Apr 22 2025, 00:00:00) [GCC 15.0.1 20250418 (Red Hat 15.0.1-0)]
Executable           /usr/bin/python
CPU                  12th Gen Intel(R) Core(TM) i9-12900H (20 cores)
Memory               62.5 GiB

Core
├☑ mne               1.9.0 (latest release)
├☑ numpy             2.2.5 (OpenBLAS 0.3.29 with 20 threads)
├☑ scipy             1.14.1
└☑ matplotlib        3.10.0 (backend=qtagg)

Numerical (optional)
├☑ sklearn           1.6.1
├☑ nibabel           5.3.2
├☑ nilearn           0.11.1
├☑ pandas            2.2.3
├☑ h5io              0.2.4
├☑ h5py              3.12.1
└☐ unavailable       numba, dipy, openmeeg, cupy

Visualization (optional)
├☑ pyvista           0.44.1 (OpenGL 4.6 (Core Profile) Mesa 25.0.4 via Mesa Intel(R) Iris(R) Xe Graphics (ADL GT2))
├☑ pyvistaqt         0.11.1
├☑ vtk               9.4.0
├☑ qtpy              2.4.2 (PyQt5=5.15.16)
├☑ pyqtgraph         0.13.7
└☐ unavailable       ipympl, mne-qt-browser, ipywidgets, trame_client, trame_server, trame_vtk, trame_vuetify

Ecosystem (optional)
├☑ mne-bids          0.16.0
├☑ edfio             0.4.8
└☐ unavailable       mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline, neo, eeglabio, mffpy, pybv

papadop, 2025-05-23:
The real difference is reading the directory in slow_mode (mandatory for "sample_audvis_raw.fif" because it does not have a directory at the end of the file) and fast mode (i.e reading the directory (list of tag headers) placed at the end of the file which is the case for  "sample_audvis_filt-0-40_raw.fif"). IMHO the result should not depend on the way of reading that directory (assuming the slow and fast directories contain the same information which is the case for "sample_audvis_filt-0-40_raw.fif").

The fact that dir_pointer is -1 for "sample_audvis_raw.fif" indicates that there is no such directory.
If you try with "sample_audvis_filt-0-40_raw.fif", you will see that dir_pointer has a non-negative value indicating the presence of that directory, which enables the fast mode.

And that's actually the reasoning why I think the current behaviour is wrong. In slow mode the first file_id header is processed while in fast mode it is skipped !

larsoner, 2025-05-23:
Ahh that makes sense @papadop ! Would you be up for making a PR to fix it?

closed 2025-06-02 by larsoner