mne-tools/mne-python #13289: ENH: make `times` parameter of `plot_evoked_joint` more versatile

sappelhoff, 2025-06-19:
[plot_evoked_joint](https://mne.tools/stable/generated/mne.viz.plot_evoked_joint.html) accepts a `times` parameter:

https://github.com/mne-tools/mne-python/blob/8df70aea97c42a58d02e1c121679e9c1b2c62869/mne/viz/evoked.py#L1818-L1822

I think the following changes would be an improvement:

- make it possible to determine the number of evenly spaced topos via `"auto"` (currently hardcoded to 5) and `"peaks"` (currently hardcoded to 3)
- make it possible to determine a window (or one window per peak) in which `"peaks"` should be found (currently hardcoded to full time window)

suggested API:

- accept a dict mapping a single str key to value(s):
- if `"auto"`, must be either:
    - an integer bigger than 1, corresponding to the number of evenly spaced topos to be plotted
    - a tuple of the form `(n, (tmin, tmax))` to plot `n` evenly spaced topos in the window `(tmin, tmax)`
- if `"peaks"`, must be either:
    - an integer bigger than 1, corresponding to the number of peaks (over full time window) to be plotted
    - a tuple of tuples, where each tuple refers to a `(tmin, tmax)` of a window in which to plot the peak of that window

For example

```python
times = {"auto": 10}  # will plot 10 evenly spaced peaks over the whole window
times = {"auto": (3, (0.1, 0.8))}  # will plot 3 evenly spaced peaks over the window from 0.1 to 0.8s

times = {"peaks": 1}  # will plot 1 single peak within the full time window
times = {"peaks": ((0, 0.1))}  # will plot single peak within 0 to 0.1s window
times = {"peaks": ((0, 0.1), (0.5, 0.8))}  # will plot two peaks within their windows

times = {"mykey": 5}  # error, only "peaks" and "auto" are accepted keys
```

From just thinking about it (haven't looked at the code yet), this should be fairly straight forward, and helpful.

WDYT?

drammock, 2025-06-23:
xref to #12857.  It would be nice if what was decided there (but never implemented yet) also worked here (or vice-versa) so that we were internally consistent.

drammock, 2026-01-02:
> Hi, I have gone through the discussion above and would like to work on this. Is it still available?

AFAIK nobody has work-in-progress on this, so feel free to open a PR @shruti423 

Please use https://github.com/mne-tools/mne-python/issues/12857#issuecomment-2375303080 as the guide for what we want to have implemented. My guess is that it makes sense / will be easier to do `evoked.plot_topomap()` and `plot_evoked_joint()` in the same PR, but that's up to you; if it seems easier to split it into 2 PRs that's fine.