mne-tools/mne-python #13294: control randomness in fit_dipole function

sappelhoff, 2025-06-24:
Based on results I am currently getting, I assume that [`mne.fit_dipole`](https://mne.tools/stable/generated/mne.fit_dipole.html#mne.fit_dipole) contains some randomness.

Can somebody pinpoint me to the source of that randomness?

Furthermore, would you agree that (as with almost all randomness in analysis code) we should be able to pass an RNG object (or `random_state`) as a user to control that randomness explicitly?

larsoner, 2025-06-24:
I don't think it used to be random. I know Syfy recently overhauled their cobyla code maybe that's the culprit and we need a seed now or something? 

In any case the first step is to figure out where the randomness is coming from

sappelhoff, 2025-07-03:
sorry @larsoner I didn't get to it yet for a "nice" comparison or investigation, but perhaps if I give you some more context, that can help you to spot whether or not there might just be an error in my reasoning.

Basically, I am trying to fit a dipole on each ICA component, this is something that is regularly done in EEGLAB, but we do not really have a workflow for it in mne.

With the code below, I am following recommendations (or hints) by @wmvanvliet in: https://mne.discourse.group/t/current-dipole-fitting-using-independent-components/6215/2?u=sappelhoff

I have two functions, one is faster because it pretends ICs are really just timepoints, and therefore the dipole fitting code source space setup only needs to run once:

```python
def fit_dipoles_on_ica_fast(ica, info, trans=None):
    # load headmodel and source space
    fs_dir = mne.datasets.fetch_fsaverage()
    bem = fs_dir / "bem" / "fsaverage-5120-5120-5120-bem-sol.fif"  # conductor model

    # for dipole fitting, need an identity covariance matrix
    cov = mne.cov.make_ad_hoc_cov(info, std=dict(eeg=1))

    # Get a dipole for each non-excluded component
    components = ica.get_components()  # (n_channels, n_components)

    # subset info to only pick good channels, corresponding to ICA fit
    sel = [
        i
        for i, (ch, typ) in enumerate(zip(info.ch_names, info.get_channel_types()))
        if ch not in info["bads"] and typ == "eeg"
    ]
    info_clean = mne.pick_info(info, sel=sel)

    # Create an Evoked object from the ICA source
    # each timepoint corresponds to one IC
    evoked_ica = mne.EvokedArray(components, info_clean)
    evoked_ica.set_eeg_reference(ref_channels="average")

    # fit dipoles on ica
    dipoles, residuals = mne.fit_dipole(
        evoked_ica, cov=cov, bem=bem, trans=trans
    )

    return dipoles, residuals
```

The other one is slower, because it sets up dipole fitting once per IC:

```python
def fit_dipoles_on_ica_slow(ica, info, trans=None):
    # load headmodel and source space
    fs_dir = mne.datasets.fetch_fsaverage()
    bem = fs_dir / "bem" / "fsaverage-5120-5120-5120-bem-sol.fif"  # conductor model

    # for dipole fitting, need an identity covariance matrix
    cov = mne.cov.make_ad_hoc_cov(info, std=dict(eeg=1))

    # Get a dipole for each non-excluded component
    components = ica.get_components()  # (n_channels, n_components)

    # subset info to only pick good channels, corresponding to ICA fit
    sel = [
        i
        for i, (ch, typ) in enumerate(zip(info.ch_names, info.get_channel_types()))
        if ch not in info["bads"] and typ == "eeg"
    ]
    info_clean = mne.pick_info(info, sel=sel)

    ncomps = components.shape[1]
    dipoles, residuals = [], []
    for icomp in range(ncomps):
        end = "\r" if icomp < ncomps - 1 else "\n"
        print(f"Fitting dipoles ({icomp + 1}/{ncomps}) ...", end=end)

        # do not work on bad components
        if icomp in ica.exclude:
            dipoles.append(None)
            residuals.append(None)
            continue

        # Create an Evoked object from the ICA source
        evoked_ica = mne.EvokedArray(components[:, [icomp]], info_clean)
        evoked_ica.set_eeg_reference(ref_channels="average")

        # fit dipole on ica
        dipole_fit, residuals_fit = mne.fit_dipole(
            evoked_ica, cov=cov, bem=bem, trans=trans
        )

        dipoles.append(dipole_fit)
        residuals.append(residuals_fit)

    return dipoles, residuals
```

Running the slow versus the fast function on the same input data, I got slightly different results ... this triggered the issue here.

Any ideas on whether there might be another mistake, unrelated to the randomness (which may not exist)?

sappelhoff, 2025-07-03:
☝ now posting this, I am wondering whether it could be because the underlying data (the ICA components) is modified when I call the slow and fast function one after the other.

I am calling `evoked_ica.set_eeg_reference(ref_channels="average")`, after all ... or perhaps fit_dipole is somehow changing the input data (unlikely, but who knows).

This would only be an explanation if `get_components` is NOT returning a copy.

sappelhoff, 2025-07-07:
> This would only be an explanation if get_components is NOT returning a copy.

modifying the data returned by `get_components`  would NOT affect any underlying data.

https://github.com/mne-tools/mne-python/blob/8f83332cac8b0cdbc28ca038ef4a66c6f876dc9c/mne/preprocessing/ica.py#L1079-L1090

larsoner, 2025-07-07:
@sappelhoff I'll push the milestone for this to the 1.11 but happy to work on it if you can make a minimal example that uses `sample`!

sappelhoff, 2025-07-08:
Thanks a lot @larsoner that explains it for me!